Stephane-D/SGDK #378: FM6 channel doesn't work in XGM2

Ganso, 2024-12-12:
When a song uses the FM6 channel, it works fine in XGM1 ("XGM_startPlay {song}") but in XGM2 ("XGM2_play {song}") that particular channel isn't played.

I've tried using YM2612_disableDAC() function before playing the song, but nothing changes. Of course, I was not playing any PCM sample while testing.

I've done all the test with [this song](https://github.com/Ganso/The-Weave/blob/master/res/Sound/Music/TheWeave_intro.vgm).

Stephane-D, 2024-12-12:
Spent a bit of time investigating the problem and i couldn't see any issue both in the XGM2 driver and in the XGM2Tool converter.
Just to be sure, did you disable DAC in the VGM file itself ? If you keep it enabled then XGM2 will follow that consign (keep DAC enable on YM chip) so FM6 will never play. If you disable DAC, XGM2 driver will automatically enables it when playing PCM and returns it to the VGM consign (disabled here) as soon PCM play end.

Stephane-D, 2024-12-12:
Should be fixed now :)

closed 2024-12-12 by Stephane-D