Stephane-D/SGDK #384: Compilation Bug Caused by 40071af commit

saintienn, 2025-01-18:
in commit [40071af](https://github.com/Stephane-D/SGDK/commit/40071af1b77053aa1bda70481b361c347faf70f5) some edit to add LEGACY_ERROR_HANDLER in sys.c causes these building errors:

C:/SGDK/bin/ld.exe: out/sega.o: in function `registersDump':
(.text.keepboot+0x24a): undefined reference to `registerState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x250): undefined reference to `registerState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x256): undefined reference to `registerState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x25c): undefined reference to `registerState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x262): undefined reference to `registerState'
C:/SGDK/bin/ld.exe: out/sega.o:(.text.keepboot+0x268): more undefined references to `registerState' follow
C:/SGDK/bin/ld.exe: out/sega.o: in function `busAddressErrorDump':
(.text.keepboot+0x2ae): undefined reference to `ext1State'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2b6): undefined reference to `addrState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2be): undefined reference to `ext2State'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2c6): undefined reference to `srState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2ce): undefined reference to `pcState'
C:/SGDK/bin/ld.exe: out/sega.o: in function `exception4WDump':
(.text.keepboot+0x2da): undefined reference to `srState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2e2): undefined reference to `pcState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2ea): undefined reference to `ext1State'
C:/SGDK/bin/ld.exe: out/sega.o: in function `exceptionDump':
(.text.keepboot+0x2f6): undefined reference to `srState'
C:/SGDK/bin/ld.exe: (.text.keepboot+0x2fe): undefined reference to `pcState'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Bus_Error':
(.text.keepboot+0x310): undefined reference to `busErrorCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Address_Error':
(.text.keepboot+0x326): undefined reference to `addressErrorCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Illegal_Instruction':
(.text.keepboot+0x33c): undefined reference to `illegalInstCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Zero_Divide':
(.text.keepboot+0x352): undefined reference to `zeroDivideCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Chk_Instruction':
(.text.keepboot+0x368): undefined reference to `chkInstCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Trapv_Instruction':
(.text.keepboot+0x37e): undefined reference to `trapvInstCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Privilege_Violation':
(.text.keepboot+0x394): undefined reference to `privilegeViolationCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Trace':
(.text.keepboot+0x3aa): undefined reference to `traceCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Line_1010_Emulation':
(.text.keepboot+0x3c0): undefined reference to `line1x1xCB'
C:/SGDK/bin/ld.exe: out/sega.o: in function `_Error_Exception':
(.text.keepboot+0x3d6): undefined reference to `errorExceptionCB'
make.exe: *** [out/rom.out] Error 1
Process terminated with status 2 (0 minute(s), 57 second(s))
25 error(s), 507 warning(s) (0 minute(s), 57 second(s))

Stephane-D, 2025-01-18:
You need to delete the sega.s file from your project folder so it will get updated with the last version from SGDK folder. be careful to preserve your custom change on this file if you have any.

closed 2025-01-18 by Stephane-D

saintienn, 2025-01-19:
But the problem is that there is no sega.s file in my project folder, should it be there? So, I will include the file

saintienn, 2025-01-19:
Thank you so much! I will change this file