Stephane-D/SGDK #390: XGM includes file metadata into the ROM

thomasgoldstein, 2025-02-27:
If a vgm or xgm (maybe wav too?) file contains metadata, that text will be included in the ROM. I think it's a waste of space, as no one will read that.
Would be nice if XGM did not include metadata into the ORM. Or maybe have the option to include it or not, if you think it's sometimes desirable. Thanks.

thomasgoldstein, 2025-03-07:
I'll look into it. Thanks.
But I think it may still be worth looking into making the SGDK support that out of the box. If you disagree, feel free to close this.

Stephane-D, 2025-03-13:
I agree with @elikfurky that it's just better to remove it from VGM if required. For instance XGM player does use them to display track information, but in game situation you may want to remove it. I could add an option in xgmtool to discard them but to be honest I prefer not having too many options in the tooling if others options are availables.

thomasgoldstein, 2025-03-13:
Up to you, don't hesitate to close this if this goes against your best judgement!

Stephane-D, 2025-03-17:
Alternatives solutions exists !

closed 2025-03-17 by Stephane-D