Stephane-D/SGDK #415: Build infrastructure issues

iratahack, 2025-04-29:
@Stephane-D , it looks like some changes have gone into the build infrastructure which are a bit strange. Some _C_ files have include files with a path starting with _**src/...**_ for example, in **src/ext/minimusic/minimus.c:**

```
#include "ext/minimusic/minimus.h"
#include "src/ext/minimusic/minimus_drv.h"
```

The second include file prefixes the path with _**src/**_ which is not needed because _**src**_ is specified as an include path. There are a few other files which do this listed below... 

src/ext/minimusic/minimus.c:#include "src/ext/minimusic/minimus_drv.h"
src/snd/xgm2.c:#include "src/snd/xgm2/drv_xgm2.h"
src/snd/pcm/snd_pcm.c:#include "src/snd/pcm/drv_pcm.h"
src/snd/pcm/snd_dpcm2.c:#include "src/snd/pcm/drv_dpcm2.h"
src/snd/pcm/snd_pcm4.c:#include "src/snd/pcm/drv_pcm4.h"
src/snd/xgm.c:#include "src/snd/drv_xgm.h"
src/snd/sound.c:#include "src/snd/drv_null.h"

Fixing these include paths causes a secondary build issue shown below:

```
m68k-elf-gcc  -DSGDK_GCC -m68000 -Wall -Wextra -Wno-array-bounds -Wno-shift-negative-value -Wno-unused-parameter -fno-builtin -fms-extensions -ffunction-sections -fdata-sections -Iinc -Iout/release -Bbin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-tree-loop-ivcanon -fomit-frame-pointer -flto -flto=auto -ffat-lto-objects -c src/vdp.c -o out/release/src/vdp.o
In file included from src/vdp.c:18:
inc/font.h:18:10: fatal error: res/libres.h: No such file or directory
   18 | #include "res/libres.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [makelib.gen:129: out/release/src/vdp.o] Error 1
```

This appears to be a dependency issue, _**out/release/res/libres.h**_ has not been created yet, although it does exist under **$(GDK)/res**. It's like a new and old environment exist at the same time.

The reason I found these issues is because they break the CMake build unless I add $(GDK) to the include paths. This of course I can do, but it doesn't address the underlined inconsistencies. 

Any ideas how to move forward?
Thanks.

werton, 2025-04-30:
> The second include file prefixes the path with src/ which is not needed because src is specified as an include path. 

judging from the several recent changes makefile changes, this is now required because the src/ folder is no longer included in include path

before:
`.INCS:= -I. -I$(INCLUDE) -I$(SRC) -I$(RES) -I$(INCLUDE_LIB) -I$(RES_LIB)`

now:
`INCS := -I. -I$(INC_DIR) -I$(RES_DIR) -I$(OUT_DIR) -isystem$(INC_LIB) -isystem$(OUT_DIR_LIB)`

iratahack, 2025-04-30:
@werton , that's what I originally thought. But if that were the case, then the include of the first file #include "ext/minimusic/minimus.h" would fail, which it doesn't.

werton, 2025-04-30:
@iratahack , This works because the compiler, as I recall, always looks for .h first in the current folder (if the included header is written in "") of .c the file where is writen #include. If the header is included in a file that is in another directory (for example, the header located in an inc/ folder includes the header located in a src/ folder), this will cause an error.

iratahack, 2025-04-30:
@werton , Ah, that explains why it works. What's the reason for the inconsistency? Some includes having the src/ prefix and others not and what causes the build error when the inconsistency is made consistent? Or should I just add the GDK directory to the include path in the CMakeLists.txt file and be done with it?

werton, 2025-04-30:
@iratahack , As I understand it, Stef wanted to improve the dependencies so that the compiler makes fewer passes when including headers, which should have a positive effect on compile time. I don't see any particular inconsistency here, for consistency you can always specify the full path (from the project root) when including headers, or note that if a header is stored in src/ and is included in a file from a different directory, you should specify the full path before the header. Alternatively, you can add not the whole $(GDK) but only $(GDK)/src to the inclusion path, this should work too, but it's still better to figure out where the dependencies are breaking.

iratahack, 2025-04-30:
@werton , ok, got it. I will add $GDK to the include path for the CMake build. Since CMake handles the generation of the dependencies, I don't see any other problems if I just make this one change.

closed 2025-05-03 by iratahack