Stephane-D/SGDK #418: gitignore "deletes" out folder from commit

werton, 2025-05-05:
Hi! I noticed that gitignore now "deletes" out folder (and all rom files) from commits. So  last sample turned out to be without a rom file. Maybe this is intended, because now there is a .bat file that allows to build all the samples, but now it turns out that the old examples have rom files, and the new ones will be without it.

Stephane-D, 2025-05-05:
out/rom.bin is specifically excluded from .gitignore but you may need to force add the file using `git add -f file` command.

werton, 2025-05-06:
Yes, but now the rom files in the Sample folder are stored in the "out/release/rom.bin" path.

Stephane-D, 2025-05-06:
Afaik both are allowed, having out/release and out/debug allow to separate both build but ultimately out/rom.bin is convenient for fast testing last produced ROM and also preserve old behavior, it's why the makefile always copy it to out/rom.bin at the end.
Maybe I will reverse that default out/release/rom.bin to just out/rom.bin as we had before (and by default it will contains the release version obviously)

werton, 2025-05-06:
Ok, thanks for the clarification:)

closed 2025-05-06 by werton