Stephane-D/SGDK #426: OBJECTS resource of rescomp: object type property of Tiled object cannot reference to an another Tiled object located in another array

werton, 2025-05-13:
For example, we have a .res file with this content
```
OBJECTS enemiesArray_1 "objects.tmx" "EnemyLayer_1" "target:object" "sortby:id" "TMX_Enemy"  

OBJECTS enemiesArray_2 "objects.tmx" "EnemyLayer_2" "target:object" "sortby:id" "TMX_Enemy"  
```
And there are `objects.tmx`, which has an object of 2 layers `EnemyLayer_1` and `EnemyLayer_2`. There are, for example, 4 `TMX_Enemy` objects, 2 on each layer.
If the `TMX_Enemy` object has an Tiled `object` type property that references another `TMX_Enemy` object on the **_same_** layer, then everything is ok and the .res file compiles normally.
If the `TMX_Enemy` object has an Tiled `object` type property that references another `TMX_Enemy` object on **_another_** layer (i.e., an object from the `enemiesArray_1` array references an object from the `enemiesArray_2` array), then compilation of .res files is failed with reference error. Like this:
```
sgdk/bin/ld.exe: out/res/objects.o: in function `enemiesArray_1_object_1`:
(.rodata+0x52): undefined reference to `enemiesArray_1_object_3`
```
It can be seen from the log that the name of the second object referenced by the first one was generated incorrectly, the prefix is from the first object, and the second part is from the id of the second object (i.e. real name should be `enemiesArray_2_object_3`). If it's not very clear, then later I'll make a minimalistic example with a .tmx file.

werton, 2025-05-13:
I've looked into the rescomp source, and it's probably an undocumented limitation, than a bug 🙂
It seems that there is no quick solution to this problem. To fix this one more step needs to be added - resolving dependencies between objects from different arrays/lines, which should be done after creating objects from all the lines in the .res file and before compilation.

Stephane-D, 2025-06-19:
I think this one is fixed with your PR allowing referencing resources from different sources, is not it ?

werton, 2025-06-19:
@Stephane-D Oh, that's right, I forgot I created the issue :) I'm closing it.