Stephane-D/SGDK #434: [samples] Tile animation issue with SGDK 2.11

ProtossGP32, 2025-06-15:
Hi all! I've been following SGDK progresses for quite some time and first of all I'd like to thank @Stephane-D for his awesome work as well as the rest of contributors, particularly to @werton for his latest project samples.

**DESCRIPTION:**
I've found a possible bug in the tile animation example, where some of the palmtrees' tiles aren't rendered. I attach some screenshots showing the issue, unfortunately I can't test it on real hardware:

- BlastEm emulator:
  - Missing tiles:
    [image]

  - Original tileset:
    [image]

  - VDP VRAM debug:
    [image]

I'm using doragasu's sgdk-docker:2.11 to compile the ROM, unfortunately I can't compile it in my local environment (I still have to solve a lot of issues regarding toolchain and environment setup...).

I don't see anything strange in the code, but if I disable anything related to the animated tilesets, then the palmtrees are properly rendered:

```c
    // load tileset of animated tiles (first frame)
    //VDP_loadTileSet(tile_anim[tileFrameIndex], bgaMap->baseTile, DMA);
    // ...
    while(TRUE) {
        // ...
        // update tileset of animated tiles
        //TilesAnimationUpdate();
    }
```

- BlastEm emulator - Tile animation disabled:
[image]

- VDP VRAM debug:
[image]

**POSSIBLE SOLUTION:**

It seems to me that the issue comes from the last 3 empty tiles in the tileset, just below the purple flower tiles; optimization during the background tileset generation is removing them, thus reducing the expected tileset length and overwriting non-animated tiles instead.

Having said that, and after re-reading the comments in the `resource.res` file, I see two possible workarounds:
- Option A: set the `TILESET` compression value to `NONE` instead of `ALL` to avoid removing the empty tiles after the placeholders:
  ```c
  //------------------------------ Tileset for background --------------------------------------------
  // This tileset contains placeholders for animated tiles and map tiles.
  // The placeholders can be any tiled blocks (I used hint labels),
  // but is important that the placeholders consist of unique non-repeating (including mirrored) tiles,
  // as this will allow using tileset optimization without worrying that the placeholders will be altered by the optimizer.
  // It is also important that the placeholder tiles are not on the same horizontal line as the empty tiles and the map tiles
  // (except for the last line of placeholder tiles), otherwise, when tile optimization is enabled,
  // the optimizer will remove duplicated tiles, and shift the placeholder tiles will shift.
  //---------------------------------------------------------------------------------------------------
  // TILESET bg_tileset "bga_tileset.png" BEST ALL
  TILESET bg_tileset "bga_tileset.png" BEST NONE
  ```
- Option B: modify the `bga_tileset.png` so there are no empty tiles between the placeholder tiles and the _static_ tiles.

Apart from that, the example is pretty straightforward and understandable! Thanks again for this!

werton, 2025-06-15:
Hi @ProtossGP32
Nice find, thanks. You are absolutely right. I didn't want to add additional "filler" tiles between placeholders and static tiles, but I missed that static tiles can also shifts due to optimization and end up in a rewritable place. I will update this sample as soon as I return from a work trip in July (if no one fix it before then)
And special thanks for such a detailed bug report, with screenshots, description and even a solution, I love people with such a fundamental approach :)

ProtossGP32, 2025-06-15:
> Hi [@ProtossGP32](https://github.com/ProtossGP32) Nice find, thanks. You are absolutely right. I didn't want to add additional "filler" tiles between placeholders and static tiles, but I missed that [...]

Thank you for your quick reply. My verbosity level on issue description comes from my job, too used to be as explicit as possible to avoid misunderstandings and keep good track of any progress through time, now I can't do it any other way 😅 .

Regarding a fix, I was thinking on opening a PR, but I'd rather be you who decides the best approach as the author of the sample 😉 .

Stephane-D, 2025-07-08:
Thanks for the PR @werton, I merged it so we can close the issue :)

closed 2025-07-08 by Stephane-D