Stephane-D/SGDK #441: Problem on sega.o: undefined reference

saintienn, 2025-07-28:
Hello Stef
I update SGDK to the latest version and when I compile my project I get this error:
C:\SGDK\bin\ld.exe: out\release\src\boot\sega.o||in function `no_user_task':|
(.text.keepboot+0x468)||undefined reference to `XGM_doVBlankProcess'|

What I can do about it?
thanks
Francis Fussiger

saintienn, 2025-07-28:
if I try to bypass the problem, deleting /src/boot/sega.s from my project, this happen: 
C:/SGDK/bin/gcc   out/release/src/boot/sega.o   -o out/release/src/boot/sega
gcc.exe: fatal error: '-fuse-linker-plugin', but liblto_plugin.dll not found
compilation terminated.

saintienn, 2025-07-29:
I'm not sure exactly what causes the bug, but the code only compiles and outputs a Binary if I delete the boot directory from my project. I was using the project template that came in the SGDK directory to include the game name in the header.

Stephane-D, 2025-08-04:
Last SGDK just removed the requirement of boot/sega.s file so now you can just delete the whole _boot_ folder and it should work.
I previously added a command to force the deletion of the _boot_ folder directly in the makefile but it could be harmful if you wanted to preserve your old rom_head.c file for instance so now I let the user delete it by himself (next SGDK official release will contains notes about it)

closed 2025-08-04 by Stephane-D