Stephane-D/SGDK #444: Z80 objects are included twice in subsequent builds

akumanatt, 2025-08-06:
Due to a recent makefile.gen change, if you have a Z80 code in the source as a.s80, a.s and a.o will be generated (instead of a.o80 previously). However, in subsequent builds, the generated a.s will also generate a second a.o in the object list. This causes a duplicate definition error on linking as ld will include two identical files as a separate thing.

https://github.com/Stephane-D/SGDK/blob/master/makefile.gen#L82-L84

Stephane-D, 2025-08-07:
Normally intermediate .s files are generated inside the out/<target> folder :
https://github.com/Stephane-D/SGDK/blob/master/makefile.gen#L232

So they should never been included as sources on next build process.
I think you have another issue here ?

akumanatt, 2025-08-10:
Yes. I double-checked the installation and turned out I forgot to install a new version after building so the new makefile isn't updated. Thanks for pointing this out and rebuilds work fine now.

closed 2025-08-10 by akumanatt