Stephane-D/SGDK #445: Cannot compile

supergameromegaclank, 2025-08-06:
I can't figure out how to solve the missing source. It could compile despite that before i tryied to add images

[image]

thomasgoldstein, 2025-08-07:
For support, you should ask on the SGDK Discord: https://discord.gg/xmnBWQS

@Stephane-D this keeps on happening. There are probably ways we could avoid people creating issues for support.
Maybe configuring a [default GitHub issue template](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository) that would have a big visible warning to ask questions the Discord, or something like that. 🤷‍♂️

Stephane-D, 2025-08-07:
@thomasgoldstein Just added issue templates, hoping that will help reducing that !

closed 2025-08-07 by Stephane-D